nordicsemi/IOS-CoreBluetooth-Mock #8: Library Evolution Support

kylewludwig, 2020-05-23:
Hi, I just using this library and I'm really glad it exists!

However, I'm getting this warning next to `import CoreBluetoothMock` using CocoaPods with version 0.8.0 when I enabled `Build Libraries for Distribution` in the project Build Settings:

`Module 'CoreBluetoothMock' was not compiled with library evolution support; using it means binary compatibility for 'Project Name' can't be guaranteed`

Would it be possible to enable `Build Libraries for Distribution` to improve ABI support for later versions? If not, no worries!

philips77, 2020-05-23:
I'll see what can be done! Thanks for the feedback.

philips77, 2020-08-03:
Hi,
I created a dummy project, where I created the following `Podfile`:
```
use_frameworks!
target 'TestKaczka' do
  platform :ios, '9.0'
  pod 'CoreBluetoothMock'
end
```
I then added
```swift
import CoreBluetoothMock
```
in one of the files and set `Build Libraries for Distribution` in the test target Build Settings.
Indeed I got the warning:
```
Module 'CoreBluetoothMock' was not compiled with library evolution support; using it means binary compatibility for 'Dummy' can't be guaranteed
```
But then, I set this flag for the `CoreBluetoothMock` target in `Pods` project and warning is gone. No changes in the library project. Are you sure you need me to change it, for example it will reset to *No* every time you do `pod update`?

I have no idea what this flag is for so I can set to anything you want until it compiles :)

kylewludwig, 2020-08-04:
@philips77 This feature would help with cross-platform support for the library when distributing an XCFramework like [this](https://medium.com/@priya_talreja/binary-frameworks-in-swift-3e6df1f27567) by preventing Swift version lock. I saw #6 and thought it'd be helpful to suggest. [This WWDC video](https://developer.apple.com/videos/play/wwdc2019/416/) explains how it works.

Basically, users of the library will need to archive each variant of the library for the target OS. Then, they can re-package it as an XCFramework and create a single project with one `CoreBluetoothMock` library (where business logic can be shared between apps on tvOS, watchOS, and iOS.)

philips77, 2020-08-04:
Fixed and released in 0.10.0. Could you verify?

closed 2020-08-04 by philips77