nordicsemi/IOS-CoreBluetooth-Mock #9: Unable to re-initialize native central manager after mock central manager

kylewludwig, 2020-06-22:
When creating a "demo mode" for an application, it'd be great to be able to allow users to toggle between a "native" and "mock" central manager with all supported features of the device and software. This way users can evaluate how a product behaves before the purchase by just downloading the app.

I'd found that I can do it with this library with
`self.centralManager = CBMCentralManagerFactory.instance(delegate: CBMCentralManagerDelegate, queue: DispatchQueue?, options: [String: Any]?, forceMock: Bool)`
by re-initializing a central manager with the `forceMock` toggled.

However, I've found this only works when the `CBMCentralManagerOptionRestoreIdentifierKey` is omitted from `options:` in the `init` method above. I can either remove this key and lose support for CoreBluetooth peripheral restoration or include a custom string for the key each time to enable it again.

Separately (and possibly related) when using Instruments to detect memory leaks, I found the native central manager was able to call `deinit` but the mock central manager is unable to call `deinit` when using a similar `centralManager` and `peripheral` implementation to the Blinky example provided (using notification central and a reference cycle between the central manager and peripherals).

philips77, 2020-06-22:
Thank you for the feedback. I'll look into it when I have some time.

kylewludwig, 2020-06-22:
Thanks in advance! See the last comment in this thread from RxBluetoothKit; it looks like a similar issue. If this is the case, it could very well be my code causing this memory leak, too. I just haven't found it yet.

https://github.com/Polidea/RxBluetoothKit/issues/264

kylewludwig, 2020-06-23:
I found the issue at this line caused the memory leak; keeping the previous `CBMCentralManagerNative` from calling `deinit`:
https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/blob/835c9785d41b824218f86c78154a61a3b3abb297/CoreBluetoothMock/Classes/CBMCentralManagerNative.swift#L41

My fix was to change the line to use a weak reference to the manager with force unwrapping to minimize changes:
`fileprivate weak var manager: CBMCentralManagerNative!`

If you'd like to make the change, feel free. I can also make a fork and formal PR though if you'd like 🙂

philips77, 2020-08-03:
Fixed.

PR #13 (2020-08-03): Version 0.8.1
This version fixed #9 and #6.
closed 2020-08-03 by philips77