nordicsemi/IOS-CoreBluetooth-Mock #10: typealias file seems to hide KVO properties

DrAma999, 2020-06-29:
Hello,
I've added as swift package dependency CoreBluetoothMock library and added as suggested the ` CoreBluetoothTypeAliases.swift ` to the project. Removed any reference to `import CoreBluetooth` in my framework project.
Everything seems ok, but I get an complire error here:
```
  let peripheralStatePublisher: AnyPublisher<PeripheralState, Never>
    
    init(_ peripheral: CBPeripheral) {
        self.cbPeripheral = peripheral
        self.peripheralStatePublisher = self.cbPeripheral.publisher(for: \.state)
            .map{ (state) -> PeripheralState in
                PeripheralState(state: state)
            }
            .share()
            .eraseToAnyPublisher()
        self.cbPeripheral.delegate = self.peripheralProxy
    }
```
What I'm trying to do is to leverage KVO and combine capabilities to create a publisher for peripheral `state` changes.
At the line where I try to get the publisher I get **Value of type 'CBPeripheral' (aka 'CBMPeripheral') has no member 'publisher'**.
I've tried also to import Foundation and Combine in the alias file, but I still get this error.
I'm using Xcode 11.5 and the 0.8.0 version of the library.

DrAma999, 2020-07-04:
After further investigation, is not due to aliases, but the issue is due to the nature protocol of the mock `CMBPeripheral`.

philips77, 2020-07-04:
Yes, the original `CBPeripgeral` is a class, while in this library, the`CBMPeripheral` is a protocol. For the same reason it's not `Hashable` or `Equatable`. 
Regarding the publish, I could add those methods to the protocol, but the other protocols are more problematic.

DrAma999, 2020-07-08:
> Yes, the original `CBPeripgeral` is a class, while in this library, the`CBMPeripheral` is a protocol. For the same reason it's not `Hashable` or `Equatable`.
> Regarding the publish, I could add those methods to the protocol, but the other protocols are more problematic.

It would be wonderful, to test that functionality I made a little trick, but I'm not liking it that much.
May I say that this project is simply amazing!. I'm trying to build a CB library , sort of Polidea RX Bluetooth but with combine and is helping me a lot in creating a good unit test coverage.
Thank you!

MrsKitty24, 2020-08-10:
> Hello,
> 
> I've added as swift package dependency CoreBluetoothMock library and added as suggested the ` CoreBluetoothTypeAliases.swift ` to the project. Removed any reference to `import CoreBluetooth` in my framework project.
> 
> Everything seems ok, but I get an complire error here:
> 
> ```swift
> 
>   let peripheralStatePublisher: AnyPublisher<PeripheralState, Never>
> 
>     
> 
>     init(_ peripheral: CBPeripheral) {
> 
>         self.cbPeripheral = peripheral
> 
>         self.peripheralStatePublisher = self.cbPeripheral.publisher(for: \.state)
> 
>             .map{ (state) -> PeripheralState in
> 
>                 PeripheralState(state: state)
> 
>             }
> 
>             .share()
> 
>             .eraseToAnyPublisher()
> 
>         self.cbPeripheral.delegate = self.peripheralProxy
> 
>     }
> 
> ```
> 
> What I'm trying to do is to leverage KVO and combine capabilities to create a publisher for peripheral `state` changes.
> 
> At the line where I try to get the publisher I get **Value of type 'CBPeripheral' (aka 'CBMPeripheral') has no member 'publisher'**.
> 
> I've tried also to import Foundation and Combine in the alias file, but I still get this error.
> 
> I'm using Xcode 11.5 and the 0.8.0 version of the library.

philips77, 2023-08-30:
@NickKibish, how is this issue related to #96? Is it fixed, or 96 solves just few cases?