nordicsemi/IOS-CoreBluetooth-Mock #11: Deadlock

DrAma999, 2020-07-30:
I've found a potential deadlock while trying to simulate a write without response. If I create a `CBCentralManager` mock that uses the main queue, the code will crash trying to dispatch synchronously on main. `CBMCentralManagerMock.swift` about line 1105 method 
```
public func writeValue(_ data: Data,
                           for characteristic: CBMCharacteristic,
                           type: CBMCharacteristicWriteType)
```
Here
```
else {
            queue.sync { //<--Here
                guard availableWriteWithoutResponseBuffer > 0 else {
                    return
                }
                availableWriteWithoutResponseBuffer -= 1
                _canSendWriteWithoutResponse = false
            }
```

philips77, 2020-07-30:
Could you prepare a PR with a fix, please?

DrAma999, 2020-07-30:
Sure, I'll do in the weekend

philips77, 2020-08-03:
Closed by #12.

closed 2020-08-03 by philips77