nordicsemi/IOS-CoreBluetooth-Mock #17: Ability to reset mock to initial state

s-hocking, 2020-08-05:
Hi there! I'm using this library to write automated tests for my BLE framework. It would be really helpful to be able to teardown the mock and simulated peripherals, and set it up again with a clean slate, to support automated testing. Is it possible to do this?

philips77, 2020-08-05:
What you may do, is to create peripherals ones with proximity `.outOfRange`, and then move them in or out of range for each test using
https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/blob/89e2e911b061ded12ed309d9745f27464299ed19/CoreBluetoothMock/Classes/CBMPeripheralSpec.swift#L189-L197
My assumption was that usually you have a single peripheral that your app is connecting to and performing different tests with it.
I will think about your feature request, but as far as I remember I did it for a purpose.

philips77, 2020-08-05:
Also, I had to switch to another project for now and unfortunately have no idea when can I implement this feature.

philips77, 2020-08-05:
But we will gladly accept a PR is you are willing to contribute!

s-hocking, 2020-08-05:
> My assumption was that usually you have a single peripheral that your app is connecting to and performing different tests with it.

That's definitely a big part of my tests! but another part is discovering and connecting to multiple devices with varying services / configuration. Your workaround of moving the peripherals in and out of range is a good idea.

I'd love to submit a PR for this - I'll see how I go for time.

s-hocking, 2020-08-06:
Another use case to consider is the practicality of having multiple test files, with multiple setup phases. With the current mock framework, each test setup needs to attempt setting up the peripherals for all other test case files - which means that test cases are not well isolated from each other.

s-hocking, 2020-08-06:
Resolved in 0.11.0 !

PR #18 (2020-08-06): Add tearDownSimulation() method
Adds the ability to reset the simulation to an initial state using `tearDownSimulation()`, for use in unit testing. Fixes #17 

Note that the test cases in `master` are currently not passing, and I wasn't sure how to fix it. As a result, the tests in this branch are not passing either. I have added calls to `tearDownSimulation()` in the existing test cases.
closed 2020-08-06 by s-hocking