nordicsemi/IOS-CoreBluetooth-Mock #25: Peripheral disconnect and teardown cause assertion failure

s-hocking, 2020-08-27:
This issue is a bit convoluted...I know exactly why it's happening, but I'm not sure what the best approach is to fix it.

I have a test case where at the end of the test I disconnect a peripheral. [This decrements `CBMPeripheralSpec.virtualConnections` inside a closure after a delay.](https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/blob/1562143edfff49513ed2cb453d7aaf2c9756e631/CoreBluetoothMock/Classes/CBMCentralManagerMock.swift#L749).

Immediately after my test finishes, my `tearDown()` method runs `CBMCentralManagerMock.tearDownSimulation()`. This sets `virtualConnections = 0`.

Due to the delay on the peripheral disconnect, the closure runs _after_ `tearDown()` has already set `virtualConnections = 0`, and it decrements `virtualConnections` so that it ends up set to `-1`.

Finally, [the `assert()` here](https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/blob/1562143edfff49513ed2cb453d7aaf2c9756e631/CoreBluetoothMock/Classes/CBMPeripheralSpecDelegate.swift#L192) runs and fails the test because `virtualConnections` is negative.

One idea I had for a fix is to run all internal closures on a private queue that can be paused and cleared during teardown. These closures would have to dispatch to the main queue (or the user-specified queue) to call back to delegates etc.

philips77, 2020-08-28:
I think your proposal makes sense. The `tearDownSimulation()` was added recently in #18 and #20 and looks like wasn't properly tested. The closure check if the manager is `. poweredOn `, but `tearDownSimulation()` [disables it](https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/blob/1562143edfff49513ed2cb453d7aaf2c9756e631/CoreBluetoothMock/Classes/CBMCentralManagerMock.swift#L165). I guess you reenable it again immediately, yes?

A simple fix would just be to check in that closure if the `CBMCentralManagerMock.managers` contains any mock manager with `self` (`CBMPeripheralMock`) in `peripherals`.

philips77, 2021-08-18:
Hopefully fixed with #61. Sorry, that it took so long. But I managed to look into it before 1st anniversary of this issue!

closed 2021-08-18 by philips77