nordicsemi/IOS-CoreBluetooth-Mock #33: Init() Unavailable

jpeters-ml, 2021-04-27:
Using Xcode Version 12.5 (12E262) and deploying to iOS 13.0, I am unable to build the CoreBluetoothMock pod. While building I get the following errors:

`/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMServiceTypes.swift:65:5: 'init()' is unavailable
/CoreBluetooth.CBAttribute:4:12: 'init()' has been explicitly marked unavailable here
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMServiceTypes.swift:72:14: 'init()' is unavailable
/CoreBluetooth.CBAttribute:4:12: 'init()' has been explicitly marked unavailable here
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMCentralManagerNative.swift:490:17: 'init()' is unavailable
/CoreBluetooth.CBPeer:4:12: 'init()' has been explicitly marked unavailable here
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMCentralManagerMock.swift:681:17: 'init()' is unavailable
/CoreBluetooth.CBPeer:4:12: 'init()' has been explicitly marked unavailable here
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMCentralManagerMock.swift:688:17: 'init()' is unavailable
/CoreBluetooth.CBPeer:4:12: 'init()' has been explicitly marked unavailable here
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMServiceTypes.swift:176:5: 'init()' is unavailable
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMServiceTypes.swift:176:5: 'init()' has been explicitly marked unavailable here (CoreBluetooth.CBAttribute)
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMServiceTypes.swift:184:5: 'init()' is unavailable
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMServiceTypes.swift:184:5: 'init()' has been explicitly marked unavailable here (CoreBluetooth.CBAttribute)
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMServiceTypes.swift:270:5: 'init()' is unavailable
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMServiceTypes.swift:270:5: 'init()' has been explicitly marked unavailable here (CoreBluetooth.CBAttribute)
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMServiceTypes.swift:276:5: 'init()' is unavailable
/Pods/CoreBluetoothMock/CoreBluetoothMock/Classes/CBMServiceTypes.swift:276:5: 'init()' has been explicitly marked unavailable here (CoreBluetooth.CBAttribute)`

macintosh-HD, 2021-04-27:
I get the same error. Looks like that Apple is enforcing the availability annotations on init's more strongly.
In the [Xcode 12.5 Release Notes](https://developer.apple.com/documentation/xcode-release-notes/xcode-12_5-release-notes) they mention the following:
> Clang now infers the availability of +new from availability annotations on -init methods. Since +new calls [[Foo alloc] init], +new isn’t available unless +init is available. (75884815)

jpeters-ml, 2021-04-27:
There's a PR that fixes this (https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/pull/31), until it's merged in I just forked that repo to continue testing.

philips77, 2021-04-29:
#31 has been merged to *develop* and I'll release it shortly.

closed 2021-04-29 by philips77