nordicsemi/IOS-CoreBluetooth-Mock #39: Unit test failing with EXC_BAD_ACCESS

ramunasjurgilas, 2021-05-17:
Some times unit test crashing with error: **Thread 4: EXC_BAD_ACCESS (code=1, address=0xeb0494cb1130)**

[image]

My temporary fix for this failure is, that running second or third time unit test passing ok. But any way would be fine if it would not crash at all.

philips77, 2021-05-20:
Thank you for reporting the issue. I'll look into it when have some time.

philips77, 2021-08-11:
This should be fixed on *develop* branch. Could you give it a spin?

ramunasjurgilas, 2021-08-12:
@philips77 Currently I stopped using IOS-CoreBluetooth-Mock and can not try it now.

philips77, 2021-08-12:
No prob. Thank you for reporting the issue. I'll close it then.