nordicsemi/IOS-CoreBluetooth-Mock #41: Library does not compile with Xcode 13 beta/iOS 15 SDK due to CoreBluetooth API changes

zwift-aebling, 2021-06-08:
A number of inverse relationships in CoreBluetooth (e.g. `CBCharacteristic.service`) have been changed from `unowned(unsafe)` to `weak`, and therefore Optional in the iOS 15 SDK.

This results in numerous compilation errors when attempting use the IOS-CoreBluetooth-Mock library:

```
~/IOS-CoreBluetooth-Mock/CoreBluetoothMock/Classes/CBMCentralManagerNative.swift:391:51: Value of optional type 'CBService?' must be unwrapped to a value of type 'CBService'
```

philips77, 2021-06-09:
I love Apple and it's approach to backwards compatibility. Thank you for letting us know! Could you, please, prepare a PR with a fix?

zwift-aebling, 2021-06-09:
Will do!

ricobeck, 2021-06-21:
Did it:
https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/pull/44

philips77, 2021-08-05:
This should now work on *develop* branch after #50, #51 and #52.

closed 2021-08-05 by philips77