nordicsemi/IOS-CoreBluetooth-Mock #42: Can't mock duplicate services with CBMPeripheralSpecDelegate?

jaylyerly, 2021-06-16:
I'm interfacing with a device that has two battery level services.  (The device has two physical batteries.). In Core Bluetooth, I can distinguish these by resolving the descriptors on the characteristics.  The value of the CBMUUIDCharacteristicFormatString descriptor is different for the two services.  I think that's the correct way to do it according to the spec.

But, I run into a problem when I try to mock this object.  In CBMPeripheralSpecDelegate, I can implement the following method to return the data for the descriptor.

 func peripheral(_ peripheral: CBMPeripheralSpec, didReceiveReadRequestFor descriptor: CBMDescriptor) -> Result<Data, Error> 

But I don't think I have enough information here to distinguish between the two services.  The UUID for both the service and the characteristic are the standard 'battery level' values (180F / 2A19).  I don't see another way to differentiate between the two. 

Am I missing something?  Is there another identifier I can use?  Nothing stood out looking through the source code.  Is there another mechanism for supplying the `value` of a descriptor?

Thanks for the insight!

philips77, 2021-06-17:
That's a good one. I didn't think of this use case. This is the only way to providing the value. Perhaps adding parent characteristic and service is a way to go.

jaylyerly, 2021-06-17:
Yeah, if there were some way to distinguish them, that'd be great.

Another option might be to add a `userInfo` dictionary to the mock objects.  Kind of an old-school Cocoa pattern, but sometimes it's really useful just to be able to add some arbitrary info to the objects as they get passed around.

I've only been using the library for a couple days, but it's working really really well for us!  Thanks for creating it!

philips77, 2021-08-20:
Hi,
Have a look at #62. I descried the changes there and added a test for a use case similar to your device.

philips77, 2021-08-23:
Released in 0.13.0.

philips77, 2021-08-23:
Let me know if something doesn't work as expected.

PR #62 (2021-08-23): Support for duplicate services
This PR fixes #42. It contains some breaking changes.

Changes:
- Callback methods in `CBMPeripheralSpecDelegate` with parameters of type `CBMService`, `CBMCharacteristic` or `CBMDescriptor` have been replaced with new ones with parameter of types `CBMServiceMock`, `CBMCharacteristicMock` and `CBMDescriptorMock`. The old methods will still be called, but user's code should be migrated to the new methods, as this may change in the future.
- The attribute passed to those methods are the original attributes specified in peripheral spec. Before, the per-client not-mo [... cut]
closed 2021-08-23 by philips77