nordicsemi/IOS-CoreBluetooth-Mock #45: retrievePeripherals(withIdentifiers:) does not work as expected

NeverwinterMoon, 2021-06-22:
When using the real CoreBluetooth, after I've connected to the peripheral once, I can use `retrievePeripherals(withIdentifiers:)` to retrieve already known peripherals on any consequent application launch. With CoreBluetoothMock - no such luck. I assumed these would be retrieved from the list of peripherals set with `CBMCentralManagerMock.simulatePeripherals`, but this sets peripherals to `private static var peripherals: [CBMPeripheralSpec] = []` and `retrievePeripherals` checks `private var peripherals: [UUID : CBMPeripheralMock] = [:]` of every manager and never finds anything on application consequent launch.

philips77, 2021-06-22:
Did you try `simulateConnection()`, like described here: https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock#peripheral-specs

I agree that a device doesn't have to be connected for that method to work. Let's say it's a feature request.

philips77, 2021-06-22:
The above mentioned method makes the peripheral available as connected peripheral. I need to check sources, i akurat forgot my implementation.

NeverwinterMoon, 2021-06-22:
@philips77 Thanks for the reply. But yes, the method you suggested only works with `retrieveConnectedPeripherals`, while in my production code I use `retrievePeripherals(withIdentifiers:)` and I wouldn't want to change it just for the sake of the mock working...

philips77, 2021-08-09:
Hi, could you review #57?

NeverwinterMoon, 2021-08-10:
@philips77 Currently vacationing, but will try to check it out this week

philips77, 2021-08-11:
I merged the PR to *develop* as I'm happy with the final result. You may check the [RetrievingDevicesTest](https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/blob/4debc47d98fa5a3e5b91a8545f047e7510f83dcd/Example/Tests/RetrievingDevicesTest.swift) file for some more details how it works.

closed 2021-08-11 by philips77