nordicsemi/IOS-CoreBluetooth-Mock #46: CBCentralManager.authorization is not supported

NeverwinterMoon, 2021-06-23:
This is something that exists since iOS 13 (so, pretty old by now) and can be used to check app's authorization even before initing CBManager.

Sadly, using the recommended approach of aliasing for CBM, I had to hide my production code behind build configuration flags in order to be able to use CBM, as I am using `CBCentralManager.authorization`

So, would be nice to have `CBCentralManager.authorization` and `CBManagerAuthorization`

iOS 13.0: https://developer.apple.com/documentation/corebluetooth/cbmanager/3180033-authorization
iOS 13.1+: https://developer.apple.com/documentation/corebluetooth/cbmanager/3377595-authorization

philips77, 2021-08-01:
Hi, could you please review #54?

philips77, 2021-08-04:
The PR has been merged to *develop*.

PR #54 (2021-08-04): Support for authorization added
This PR fixes #46.

However, it implement this feature some **breaking changes** had to be made:
- `CBMCentralManager` was refactored from a *protocol* to a *class* with internal init. This was required so the type methods (`CBMCentralManager.authorization` and `CBMCentralManager.supports(:)`) could be called on the main manager without the need of specifying mock or native manager. 

    **Note:** The behavior of those methods is not trivial. If simulated values were set in `CBMCentralManagerMock.simulateAuthorization(:)` and `CBMCentralManagerMock.simulateFeat [... cut]
closed 2021-08-09 by philips77