nordicsemi/IOS-CoreBluetooth-Mock #68: peripheral.discoverServices with empty list []

kscheff, 2022-08-03:
When using the mock the call to `.discoverServices` returns with am empty list.

Expected: the call should return all `mockServices` when supplying an empty list `([])`.

Documentation to this call from Apple states:
> Discussion
> 
> You can provide an array of CBUUID objects — representing service UUIDs — in the serviceUUIDs parameter. When you do, the peripheral returns only the services of the peripheral that match the provided UUIDs.
>

philips77, 2023-02-14:
Shouldn't you pass `nil` instead of `[]` to get all services?

kscheff, 2023-02-14:
The goal was to have it compatible to CoreBluetooth with minimal changes. CB takes an empty list [] as input and delivers an unfiltered result.

philips77, 2023-02-15:
I confirm. An empty `[]` list behaves the same way as `nil`. I did not know that. Thanks for reporting.

PR #70 (2023-02-15): 2nd try to fix
The fix #68 was on the wrong line, it need to be on the first filter since the ?? will not trigger since the array is unwrapped.