nordicsemi/IOS-CoreBluetooth-Mock #78: Simulated peripheral name always nil

lumo2707, 2023-01-21:
When I am trying to simulate peripherals the name of the retrieved device is always nil.

What I did so far:

1. Create a mock peripheral

```      
 let mockGateway1 = CBMPeripheralSpec
                .simulatePeripheral(proximity: .near)
                .advertising(
                    advertisementData: [
                        CBMAdvertisementDataLocalNameKey    : "mock1",
                        CBMAdvertisementDataServiceUUIDsKey : [CBMUUID(string: "f9505467-a5c5-42c2-95c8-0ec77aeXXXX")],
                        CBMAdvertisementDataIsConnectable   : true as NSNumber
                    ],
                    withInterval: 0.250,
                    alsoWhenConnected: false)
                .connectable(
                    name: "mock1",
                    services: [],
                    delegate: nil,
                    connectionInterval: 0.150,
                    mtu: 23)
                .build()
```

2. Call `simulatePeripherals` and `simulateInitalState:

```
            CBMCentralManagerMock.simulatePeripherals([mockGateway1])
            CBMCentralManagerMock.simulateInitialState(.poweredOn)
```

3. Then in `didDiscover` delegate I got one peripheral, but its name is always nil:

[image]

What did I wrong?

philips77, 2023-01-22:
Hmm.. i need to check that. 
Your still can get the name from advertising data map with `CBMAdvertisementDataLocalNameKey` key, that should work.

philips77, 2023-02-14:
OK, I managed to replicate the issue. Sorry for the long delay.

lumo2707, 2023-02-14:
No problem, in the meantime I use the `CBMAdvertisementDataLocalNameKey` as workaround.

PR #80 (2023-02-15): Bug fix: Returning peripheral name fixed
This PR fixes #78.

After the device is scanned, the `peripheral.name` returns the local name from the advertising packet under `CBMAdvertisementDataLocalNameKey`. From first connection, when the iOS discovers services and reads the *Device Name* characteristic, that name is returned and cached.
closed 2023-02-15 by philips77