nordicsemi/IOS-CoreBluetooth-Mock #95: How to start a real DFU when using this library?

everlof, 2023-03-22:
Hello,

I've integrated this library and it works great. However, our part of the code that interfaces with the DFU-library from Nordic Semiconductors is a bit problematic.

After I've created a `DFUServiceInitiator` and gonna call `start(target: target)` (https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/d79e33c7eed7f67454578ff08462ea92e97dc330/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L465), I can't, because there's no way to access the real peripheral from the `CBMPeripheral`.

Any hints on how to resolve this?

philips77, 2023-03-22:
The dfu lib should be split into 2/3 modules. Transport should be separate from the implementation. We have it in the backlog.

For now I recommend cloning the library and adding as development package and migrating to Core Bluetooth Mock.

everlof, 2023-03-23:
Okey, feels like it would be a bit easier to just expose the underlying `CBPeripheral` in the mock peripherals? At least until you've done the ticket in the backlog.

How do you use this library internally? I mean, how do you self get around this issue?

philips77, 2023-03-23:
> How do you use this library internally? I mean, how do you self get around this issue?

Simple. The DFU library was created before CBM and it's it not using it. Neither the nRF DFU app is. We test on real devices doing normal updates with [this](https://github.com/NordicSemiconductor/IOS-DFU-Library/tree/main/Example) sample.

closed 2023-06-05 by everlof