nordicsemi/IOS-CoreBluetooth-Mock #97: Service discovery not firing

brokkc1, 2023-06-21:
I am having an issue with service discovery, using version `0.16.1` via Swift package manager.

After connecting to our physical device, and issuing `peripheral.discoverServices([])` our delegate is not receiving a `didDiscoverCharacteristicsFor` callback.

Stepping through the code I have found when we set our delegate it seems that inside `CBMCentralManagerNative.swift` our delegate is thrown away and a wrapper delegate (that has no reference to ours) is used in place.

```
    public weak var delegate: CBMPeripheralDelegate? {
        didSet {
            if let _ = delegate {
                // We need to hold a strong reference to the wrapper, otherwise
                // it would be immediately deallocated.
                wrapper = CBPeripheralDelegateWrapper(self)
                peripheral.delegate = wrapper
            } else {
                wrapper = nil
                peripheral.delegate = nil
            }
        }
    }
```

I can see `didDiscoverServices` is being called on the `CBPeripheralDelegateWrapper`, but not being passed down to our delegate.

```
func peripheral(_ peripheral: CBPeripheral,
                        didDiscoverServices error: Error?) {
            smartCopy(peripheral.services)
            impl.delegate?.peripheral(impl, didDiscoverServices: error)
}
```

I set the delegate via the `CBPeripheral` we receive during `didDiscover` phase. Everything works perfectly when mocking, seems like an issue with the native flow only.

philips77, 2023-06-21:
Hello, thanks for the feedback.

> our delegate is thrown away

It is not thrown away, it's stored as `delegate`. Jus the `didSet` implementation doesn't touch it, just checks for presence. This delegate is then called in the code you provided: `impl.delegate?...`.

Try changing:
```swift
peripheral.discoverServices([])
```
to
```swift
peripheral.discoverServices(nil)
```

brokkc1, 2023-06-21:
Hi Phillip, Thanks for the quick response,

I've tried changing `peripheral.discoverServices([])` to `peripheral.discoverServices(nil)` still no luck sadly.

But I do understand what you mean, by our delegate is set to `impl.delegate` I can see that now thank you. I'll keep investigating it must be a config issue.

brokkc1, 2023-06-21:
The issue was caused by our delegate being cleaned, keeping an internal reference stopped this from happening.

closed 2023-06-21 by brokkc1