nordicsemi/IOS-CoreBluetooth-Mock #98: Can't read out values after DFU

everlof, 2023-06-26:
Hello,

I've integrated this framework and it works pretty well. However, when we reconnect after a DFU, we can connect once, then at next disconnect we don't get callbacks for the for the `didUpdateValueFor` after we've called `readValue(for characteristic`.

If I just restart the app, it works fine and connect. Do you have any hint of what to look for and how to debug this issue?

philips77, 2023-06-28:
Hi, thank you for investigation.
Do you have the issue when using mock or native variant of `CBMCentralManager`?

everlof, 2023-06-28:
I've got the issue when using `CBMCentralManagerNative`

everlof, 2023-06-28:
With the risk of repeating myself, it seems like it's not 100% correct to cache the services? I mean, since if I run purely on the `CBCentralManager`, the `CBPeripheral` received in `didConnect` doesn't have any services when it reconnects.

everlof, 2023-06-28:
Perhaps this is an even more sensible solution?

```
diff --git a/CoreBluetoothMock/CBMCentralManagerNative.swift b/CoreBluetoothMock/CBMCentralManagerNative.swift
index 761a95d..8439bdf 100644
--- a/CoreBluetoothMock/CBMCentralManagerNative.swift
+++ b/CoreBluetoothMock/CBMCentralManagerNative.swift
@@ -89,6 +89,7 @@ public class CBMCentralManagerNative: CBMCentralManager {
             manager.delegate?.centralManager(manager,
                                              didDisconnectPeripheral: getPeripheral(peripheral),
                                              error: error)
+            removePeripheral(peripheral)
         }
         
         #if !os(macOS)
@@ -120,6 +121,10 @@ public class CBMCentralManagerNative: CBMCentralManager {
             manager.peripherals[peripheral.identifier] = p
             return p
         }
+
+        private func removePeripheral(_ peripheral: CBPeripheral) {
+            manager.peripherals[peripheral.identifier] = nil
+        }
     }
     
     private class CBMCentralManagerDelegateWrapperWithRestoration: CBMCentralManagerDelegateWrapper {
```

philips77, 2023-06-30:
Thanks for the diff. I hope I'll find some time for this issue before my vacations. If not, it will never to wait few weeks.

everlof, 2023-09-08:
Any updates regarding this?

philips77, 2023-09-08:
Hi, sorry. I didn't have time to look into this library yet. @NickKibish has released version 0.17.0 but with other changes.
I hope me or Nick will find some time this month.

everlof, 2023-11-06:
Hello! Just wondering if anything's done regarding this? Or it's just postponed?

philips77, 2023-11-07:
Hello, we're busy with other tasks. This library is waiting for its turn. Sorry for the delay. Could you propose a change in a PR with a fix, that would be way faster.

everlof, 2023-11-08:
Absolutely, I just asked since you hinted that it might be within a month.

I've created an MR: https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/pull/102

philips77, 2024-02-19:
Merged and released as 0.18.0.

closed 2024-02-19 by philips77