nordicsemi/IOS-CoreBluetooth-Mock #113: Reconnection on disconnection when `forceMock == false` does not work due

nonameplum, 2025-01-14:
Hey,

The problem is that when the delegate call `central manager (_:diddisconnectperipheral:error:)` is handled, we would like to use it to automatically "reconnect" by calling `centralManager.connect(peripheral)` since attempts to connect to a peripheral don’t time out.

But it does not work, as after digging into the issue, the underlying `CBPeripheral`s are stored by the `CBMCentralManagerNative` which is removed on [`didDisconnectPeripheral`](https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/blob/b522df5fc0eae2a4ccea1bd8dc732265003a48a3/CoreBluetoothMock/CBMCentralManagerNative.swift#L93) so when `connect` is called it is blocked the [if statement](https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/blob/b522df5fc0eae2a4ccea1bd8dc732265003a48a3/CoreBluetoothMock/CBMCentralManagerNative.swift#L229) as the peripheral is no longer hold in the dictionary.

Potentially a quick fix could be:
```swift
func centralManager(_ central: CBCentralManager,
                    didDisconnectPeripheral peripheral: CBPeripheral,
                    error: Error?) {
    manager.delegate?.centralManager(manager,
                                     didDisconnectPeripheral: getPeripheral(peripheral),
                                     error: error)
    // Check if the error is present otherwise the disconnection was explicit
    if error == nil {
        removePeripheral(peripheral)
    }
}
```
But I guess, the biggest issue is that `CBPeripherals` are strongly hold by `CBMCentralManagerNative` in overall.

philips77, 2025-01-15:
Hi,
Thank you for creating the issue. Great catch.
How about first getting the reference to the peripheral, removing it, and calling the delegate:
```swift
func centralManager(_ central: CBCentralManager,
                    didDisconnectPeripheral peripheral: CBPeripheral,
                    error: Error?) {
    let p = getPeripheral(peripheral)
    removePeripheral(peripheral)
    manager.delegate?.centralManager(manager,
                                     didDisconnectPeripheral: p,
                                     error: error)
}
```

Unfortunately, I don't have time to look into it now, but I hope I can get back soon.

nonameplum, 2025-01-15:
This way it won't help unfortunately, because `connect` is called on the central manager later and the same [`if statement`](https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/blob/b522df5fc0eae2a4ccea1bd8dc732265003a48a3/CoreBluetoothMock/CBMCentralManagerNative.swift#L229) will not allow proceeding as peripheral will be removed from the dictionary at this point.

philips77, 2025-01-15:
Right. It should not remove the peripheral, or the `connect`. I don't remember why it does so. I need to spend some time recalling the project.

nathandud, 2025-03-11:
Removing the peripheral was added in #102 in response to #98, in which a peripheral is removed so that it doesn't hold onto all the services/characteristics after a disconnect.

philips77, 2025-03-11:
I think I'll clear the services instead of removing the peripheral. Native Central Manager allows to reconnect using the same `CBPeripheral` instance, it's just a matter of clearing the service cache.

I look into the initial proposal from #98.

PR #119 (2025-03-11): Bugfix: Reconnection
This PR fixes #113.

As stated in https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/issues/113#issuecomment-2713188259, the solution accepted in #102 introduced a bug preventing from reconnecting to a peripheral using native Central Manager. The peripheral instance was removed from managed peripherals on disconnection and reconnection was failing.

In https://github.com/NordicSemiconductor/IOS-CoreBluetooth-Mock/issues/98#issuecomment-1609597594 a different approach was also proposed, where the mock services were cleared. This PR applies this solution, but maki [... cut]
closed 2025-03-11 by philips77