nordicsemi/IOS-CoreBluetooth-Mock #132: `didDisconnectPeripheral` delegate method not called when using native implementation.

AdamUncao, 2025-05-06:
# Problem
The `didDisconnectPeripheral` delegate method is not being called when using the following `CBCentralManagerFactory` method to create a native central:
``` Swift
public static func instance(delegate: CBMCentralManagerDelegate?,
                                queue: DispatchQueue?,
                                forceMock: Bool = false) -> CBMCentralManager
```

However, using the factory method that includes the options parameter it works correctly:
``` Swift
// CBCentralManagerFactory
public static func instance(delegate: CBMCentralManagerDelegate?,
                                queue: DispatchQueue?,
                                options: [String : Any]?,
                                forceMock: Bool = false) -> CBMCentralManager
```

This regression appears to have been introduced in #127 

## Cause

I believe the issue stems from the fact that different delegate wrappers are being created when initializing `CBMCentralManagerNative`. When initializing `CBMCentralManagerNative` with options the code creates a version specific wrapper that all implement the appropriate `CoreBluetooth` disconnection delegate methods.

``` Swift
if iOS17 {
            self.wrapper = restoration ?
                CBMCentralManagerDelegateWrapperIOS17WithRestoration(self) :
                CBMCentralManagerDelegateWrapperIOS17(self)
        } else {
            self.wrapper = restoration ?
                CBMCentralManagerDelegateWrapperIOS12WithRestoration(self) :
                CBMCentralManagerDelegateWrapperIOS12(self)
        }
```

However, when initializing without options an instance of `CBMCentralManagerDelegateWrapper` is used. This wrapper does not implement any `CoreBluetooth` disconnection delegate methods and thus disconnection is not forwarded to the users delegate.

``` Swift
        // When this method is implemented, the other variant with
        // timestamp and isReconnecting flag isn't. However, the latter one
        // was added in iOS 17, so we need to selecrt the implementation
        // based on the OS version. Check out below.
        
        // func centralManager(_ central: CBCentralManager,
        //                     didDisconnectPeripheral peripheral: CBPeripheral,
        //                     error: Error?) {
        //     let p = getPeripheral(peripheral)
        //     p.mockServices = nil
        //     manager.delegate?.centralManager(manager,
        //                                      didDisconnectPeripheral: p,
        //                                      error: error)
        // }
```

# Proposed Solution
- Create another wrapper for iOS 5+ that supports the delegate method.
- Update the `CBMCentralManagerNative` init to create the appropriate wrapper.

philips77, 2025-05-07:
You are right. My bad. I'll fix asap. The same constructor should be called no matter if you add this parameter, or not.

philips77, 2025-05-07:
Could you have a look at my proposal? If no issues, I'll merge tomorrow and release a fix.

closed 2025-05-09 by philips77

philips77, 2025-05-09:
Released as 1.0.1! Enjoy and thank you again for pointing the bug. Please, let us know if you have more feedback.